RetroFlag/retroflag-picase #4: Suggest adding safe termination of emulationstation to save metadata

Brandoskey, 2018-04-22:
I've modified your script in retropie so emulationstation is gracefully terminated and metadata is saved

#functions that handle button events
def when_pressed():
  led.blink(.2,.2)
  os.system("sudo killall emulationstation && sleep 5s && sudo shutdown -h now")
def when_released():
  led.on()
def reboot(): 
  os.system("sudo killall emulationstation && sleep 5s && sudo reboot")

This was at the suggestion of an amazon user named Crysknife007

There may be a better way to accomplish this but this method definitely works. Obviously something more universal that does the same if you're using attractmode or some other front end would also be ideal.

crcerror, 2018-04-26:
This is a bad method!

Use python handler
import subprocess
output = subprocess.check_output(['./multi_switch.sh', '--es-pid'])

to get PID from ES binary if value>0 then EmulationStation is running
and now use os.system("./multi_switch.sh --es-shutdown")
to prober close system... and save metadata in every usecase
If value 0 is returnd use regular shutdown 
os.system("sudo shutdown -h now")

multi_switch.sh can be used from my github here
https://github.com/crcerror/ES-generic-shutdown

This is the swiss knife for ES shutdowns.
Use as you want, but give credits to the original author!

Brandoskey, 2018-04-26:
I don't doubt you're correct but could you elaborate on why this is a bad method?

I'm all for a better method of colurse!

Brandoskey, 2018-04-26:
That makes sense