RetroFlag/retroflag-picase #9: Uninstall

danielkoellmann, 2018-05-11:
Hi,

I'm looking for a way to completely uninstall this script? I installed without noticing that I don't have the 2nd generation RetroFlag case with the shutdown switch on the PCB. Thanks in advance. 

Kind regards, Daniel

seamonkey420, 2018-05-15:
which version did you install? i can prob assist. i learned a lot about scripting in the last month after getting a proper script for OSMC and the new case :)

danielkoellmann, 2018-05-16:
Hi,

I didn't know that various versions exist, so I can't really tell. 

I simply installed the script via "wget -O - "https://raw.githubusercontent.com/RetroFlag/retroflag-picase/master/install.sh" | sudo bash". 

I just wished there was a "remove.sh" included to revert it back to stock settings. 

Thanks!

Balveron, 2018-06-16:
Hi, 
there is still need of some advice here how to uninstall the skript... after ive installed the skript my pi wont work properly, it has lags and some options on raspi-config, like overclocking wont work anymore.
Even if its not that Script causing those errors, some advice to uninstall it would be nice.

derreck503, 2018-08-01:
@Balveron @danielkoellmann 

To uninstall follow these steps:
1. ssh into your retropie or press f4 to get to the retropie terminal
2. Type and run `sudo apt-get ---purge remove python3-gpiozero`
3. Type and run `cd /opt/RetroFlag`
4. Type and run `sudo rm SafeShutdown.py`
5. Type and run`cd /etc`
6. Type and run `sudo nano SafeShutdown.py`
7. Remove only the follow line of code: ` /opt/RetroFlag/SafeShutdown.py &`
8. Hold ctrl and press x and then type yes to save changes
9. Reboot and your set!

rene650, 2019-02-10:
Hi,
can you please tell me how to do this on Recalbox?

Kind regards, Rene

waldo68, 2019-07-26:
Would this work if I installed the GPi case script accidentally?

rrrrrrkevin, 2019-07-30:
I just accidentally installed the GPi script on my MegaPi case. I also had the switch on the "off" position. The only thing the switch does is turn shutdown my MegaPi when I turn the switch to the "on" position. I tried just running the correct installation script and nothing changed.

waldo68, 2019-07-30:
I contacted the vendor I used for my nespi+ case. These are the instructions I received. Haven't had a chance to evaluate the result yet. 
1. sudo rm -rf /opt/RetroFlag
2. sudo sed -i -e "s/^sudo python3.*//g" /etc/rc.local
3. reboot

lghtanddrk, 2019-10-28:
@derreck503 "To uninstall follow these steps:

    ssh into your retropie or press f4 to get to the retropie terminal
    Type and run sudo apt-get ---purge remove python3-gpiozero
    Type and run cd /opt/RetroFlag
    Type and run sudo rm SafeShutdown.py
    Type and runcd /etc
    Type and run sudo nano rc.local
    Remove only the follow line of code: /opt/RetroFlag/SafeShutdown.py &
    Hold ctrl and press x and then type yes to save changes
    Reboot and your set!"

I have a RetroFlag Super Nes case with the shutdown enabled. I am removing my pi and installing it in a cabinet. So this uninstall process will work for me too so that I can go back to the normal menu driven shutdown?

lghtanddrk, 2019-10-28:
@waldo68  "I contacted the vendor I used for my nespi+ case. These are the instructions I received. Haven't had a chance to evaluate the result yet.

    sudo rm -rf /opt/RetroFlag
    sudo sed -i -e "s/^sudo python3.*//g" /etc/rc.local
    reboot"

Did you ever try this to see if it works? Did it completely uninstall the RetroFlag script and restore it to the original shutdown settings?

lghtanddrk, 2019-10-28:
@danielkoellmann Were you successful in removing the RetroFlag shutdown script and restoring it back to the original settings? If so, what method did you use?

waldo68, 2019-10-28:
Honestly haven't evaluated. Once summer was over the kids have been too
busy to play on there and I have been wrapped up in other projects (like
Halloween costumes!).
Hoping to get cracking on it soon though.

crcerror, 2020-01-03:
I've created a uninstall script

For RetroPie
`wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/uninstall_all.sh" | sudo bash`

For RecalBox and BATOCERA
`wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/uninstall_all.sh" | bash`

yeeshkul, 2020-07-27:
How to uninstall it from RecalBox please?

yeeshkul, 2020-07-27:
I found the answer - for Recalbox
Use PuTTy
1. Make the files editable: mount -o remount, rw /
2. Edit file etc/init.d/S99RetroFlag - remove the unwanted line 
nano etc/init.d/S99RetroFlag
2. Delete script files from opt/RetroFlag
rm /opt/RetroFlag/file_name

mccorkled, 2022-07-04:
> I've created a uninstall script
> 
> For RetroPie `wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/uninstall_all.sh" | sudo bash`
> 
> For RecalBox and BATOCERA `wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/uninstall_all.sh" | bash`
> 
> It removes scripts, cleans autostart and disables uart

In RetroPie I get the following message:

`Try to delete dir: /opt/RetroFlag
/opt/RetroFlag: removed successful!

Try to remove autostart-feature
Autoremove failed!
Sorry`

I’m not sure what the autostart feature is. Perhaps it’s not uninstalling it because it didn’t exist?

crcerror, 2022-07-16:
Yes the message is thrown out if `grep -q "^sudo python3 $SCRIPTDIR" "/etc/rc.local` fails.
There are meanwhile 1000 methods to get these scripts running.

But if the remove of /opt/RetroFlag was successful then the script should not run at all.

mccorkled, 2022-07-16:
I ended up going with a fresh install of everything. I installed your script and it seems like it’s shutting down properly but the unit will not power on until the power cable is unplugged and reconnected. 

Dustin