RetroFlag/retroflag-picase #18: Safe Shutdown Not Working - PCB switch could be wired back to front!

RobotLimeLtd, 2018-06-26:
I put the Switch to "on" and couldn't get the buttons to work.  Flipped the switch over to "off" and now it works perfectly!  So, a really simple thing for people to check if the scripts don't appear to work.

ESMP, 2018-07-20:
Great! This worked for me too. What confused me at first was that the manual said the board had the switch off by default but when I looked at it, it was "on", which apparently means "off" to them haha. Thanks for the tip!

war3zlod3r, 2018-09-08:
I'm having the same issue as you I'd love to know if you find or found a resolution, my switch is set to on and the power/reset do nothing other than turn the system on, I have it off and it just cuts the power.

war3zlod3r, 2018-09-08:
just tested on my Raspberry Pi 3B and it works, but however will not work on my 3B+

war3zlod3r, 2018-09-09:
Oddly I purchased 2 pi 3B+ models, one US style case and one EU/JP Style case, the US style worked flawlessly and easily I'm only having the issue on my 3B+ with the EU/JP case, when I use my 3B it works so I know its not the PCB and I know having the PCB switch set to on is the right position on my board, does this mean I have bad GPIO pins?

tmakriya, 2018-09-26:
Hi I am also having an issue with the safe shut down script.

I have a pi 3 b+
The superpi Eu/jp style case

I checked PCB switch in both positions, and it's wired correctly to work in the on position.

So the script appeared to install correctly from the terminal.  The problem is when I reset or switch the power to off, it exits emulation station and goes to the terminal, however from there it does not run the script, it just shuts down.

Wondering if anyone else has had this problem and been able to fix it.

Thanks,

tmakriya, 2018-09-27:
Hi,

Incase anyone else is having the same problem, I solved my issue.

I was using a pre built image.  I found that I had to go into the /boot/cmdline.txt file and edit the following:

Change "console=tty3" to "console=tty1"
And I had to remove "quiet"

Now if functions perfectly.  Maybe even before it was functioning correctly but It was hiding the script when running.

dovik, 2018-10-16:
@tmakriya Thanks for the solution.

Is your LED working with this hack? Mine is not.

vice350z, 2019-06-25:
Bad pins for me here too on my one 3B+. Tested on another Pi3B+ and a 3 and it works fine.