RetroFlag/retroflag-picase #26: Power button does not trigger shutdown script

codecomp, 2018-08-23:
I have just installed a 3b+ in a new nesipi+ case. Setup a new install of retropi and installed the shutdown and reset script. while the reset button successfully triggers the reset script, the power button doesn't trigger the shutdown script.

If I shutdown the pi via terminal then turn off, and the back on the power button it will trigger the startup. So the button and LED do both work.

daboodamonk, 2018-12-11:
I have exactly the same issue with the MegaPi case.
Was there a fix to this?

codecomp, 2018-12-11:
It was assumed to be a hardware issue after much debugging and testing. Issued a RMA with the place I bought it, though I never got around to posting it back because I'm a lazy bastard.

Fozziefett, 2018-12-31:
@daboodamonk, having same problem with MegaPi. @codecomp, considering we are all seeing the same kind of problem I assume there is a script problem as we have different cases. I can switch my pi to the NesPi and see if I see the same behavior with the script. Need to do some more research and determine how to debug.

codecomp, 2018-12-31:
I find it unlikely to be a script issue. I wrote custom scripts to monitor the current state of the IO when using the power and reset switches. They did not function in a way that would have worked. This was when using custom scripts and the default scripts. Tested it with a totally fresh install as well just to be sure.

I never got around to sending the old case off but am currently looking at getting a new case from another source so I'll see if that has the same issue. So Id that one works I'll look into what the cause could be hardware wise. Could just be a iffy toggle switch.

Does the MegaPi have a latch switch like the nes or a toggle switch?

Fozziefett, 2019-01-01:
Thanks for the response, Chris. Sounds like you have looked at this a lot
longer than I have. The board has an on/off toggle and it slides back and
forth correctly as you slide the on/off left to right.

The thing that is odd is that sliding the power toggle off and then
resetting will have the result of a power off. So the Pi seems aware of the
off state when resetting, but just isn't picking up the switch off event. I
can try another MegaPi case but when I saw one or two MegaPi owners
reporting the same issue, I suspected it was a script issue vs. an issue
with the devices. Perhaps the switch is toggle.

I'll keep you posted as a try a few things.

- Mike

codecomp, 2019-01-01:
When you slide the toggle to off and reset does the pi run the shutdown command fully, as in, can you turn it back on after the shutdown? that was the core issue with the nespi+ while a shutdown could be forced it would not repower up due to the hardware.

One quick way to check without messing around is to modify the safeshutdown.py code and comment out the definitions and replace them with prints. When i did that i was able to easily check that the power off command never fired properly for me. It was then i started checking the IO on the PI to see if this was being received on a wrong pin or some such.

Does the MegaPi use the same installer? As far as I'm aware this was all write a good year before that case even came out, so if it does it sounds like some bad choices were made to keep the code the same for what should be different pieces of hardware. Someone rewrote the installer to make it work better and even differentiate hardware (https://github.com/crcerror/retroflag-picase/) so I'm amazed they didn't do something like that themselves. Might be worth a shit, though it never worked for me.

daboodamonk, 2019-01-01:
I bought another MegaPi case and still have the issue so, I don't think its hardware related.
With the safe shutdown switch set to off (its default out of the box setting), I am able to power off the Pi with the Off button. However, this isn't a safe shutdown but rather a forced shutdown (similar to yanking the power off).

codecomp, 2019-01-01:
That defiantly sounds like a slightly different problem to the one I have.

Where did you buy it from? Same place as last time within a close span between them? As it could possibly be an issue with a batch of the products.

Fozziefett, 2019-01-01:
> I’m using the MegaPi with modified version of the retropie shutdown
> script. The on/off portion of the script works perfectly fine as intended
> for me. Provides a safe shutdown. I’m using 10/09/2018 raspbian lite.

OK, perhaps I am not using the modified script. Can you point me to the
script you are using or did you modify yourself?

keilmillerjr, 2019-01-01:
@Fozziefett Here’s a video of the script and it working on the megapixel case.

I’m also going to reference [powerenPin? No documentation. #41](https://github.com/RetroFlag/retroflag-picase/issues/41) because @RetroFlag should add documentation. Many people wish to modify this script and have no idea what pins do what.

https://youtu.be/-XvnSVDr6Hw

Fozziefett, 2019-01-02:
Thanks for the help, will check it out!

codecomp, 2019-01-02:
From my experimentation and remmebring from months ago.

Power is always running through to the internal case hardware
Pressing the power on passes power to the pi to power it on
On start up the pi then passes a signal out to the power light and starts watching for io pin signals

When you power off in normal mode the internal case hardware will cut power to the board fully

If you set the internal switch to on for safe reset the board should send a signal to the IO pin that's being watched by the startup script.
This should trigger the safe shutdown.
After this somehow the case should know the power down has cycled fully and cut power to the pi so it can be turned back on. 

It's this last part that doesn't work for me. The board seems to keeps passing power to the pi despite the pi shutdown script completing, So the pi can't be turned back on after the shutdown finishes.

I never got a chance to look into the hardware to see how it is determining the showdown is complete as hardware is not my forte. Nothing seemed blown or loose.

codecomp, 2019-01-02:
I should clarify, the power button does not trigger the shutdown script for me as mentioned in the first comment. If i manual run the shutdown, or set the reset button to trigger the shutdown script, the hardware keeps passing power to the board.

CookiePLMonster, 2019-01-02:
Basing on my research around this topic with a MegaPi case:

- ON/OFF switch state is not exposed to GPIO ports, so Pi is effectively unaware of it
- ON/OFF switch only controls whether the power should be shut down or not - in both cases Pi upon pressing Reset Pi receives same signal (that is, what stock retroflag's script considers "Reset")
- If switch is set to OFF and pin 4 (`powerenPin`) is set to off after being set to on, case cuts power. `powerenPin` is effectively used as a "I am still working" marker.

For the record, everything works fine for me. I am using a modified Multi Switch Shutdown script, but it effectively works identical as far as IO actions are concerned. I'll edit this issue in a bit with a snip of my Python script so it can be compared if needed.

daboodamonk, 2019-01-02:
OK this works for me now but its the oddest thing as to how it started working.

So, I took the case apart and removed the connector from the case to the GPIO pins on the Pi (3B+)
Then, reattached the connector, assembled the case and tested it out out again.
Nada, same deal. Reset button works fine but turning the power switch to off does nothing i.e. doesn't trigger the safe shutdown script.
So, I took the case apart again and just for kicks, removed the connector, flipped it and connected it to the GPIO pins.
I didn't expect this to work and as predicted, it didn't work (it didn't even power on the Pi as expected).
Then for the last time, I took the case apart and connected everything the way it should be.
After that to my utmost surprise, it works now! The power switch when turned to off now triggers the safe shutdown script.

I have no idea what the heck happened. I feel like Forrest Gump bumbling my way into a "solution".
Maybe this might work for you @Fozziefett ?
Seems like such a dumb "solution" but it somehow worked for me. 

P.S. before I did all this, I:
- Deleted the SafeShutdown.py file from the /op/RetroFlag folder and ran a modifed script located here:
https://github.com/crcerror/retroflag-picase
It didnt help as I was having the same issues as before so..
- I Deleted the SafeShutdown.py file from the /op/RetroFlag folder again and ran the original RetroFlag script:
https://github.com/RetroFlag/retroflag-picase
Again, that didnt help and I still had the same issues.
Thats when I started taking the case apart and fiddling around with the connector to the GPIO pins

codecomp, 2019-01-02:
I'd have put money on that blowing your Pi rather than it magically fixing the case.

Please go buy some lottery tickets

daboodamonk, 2019-01-07:
So when the PCB SAFE SHUTDOWN is On and I switch the power button off nothing happens.
When I shutdown the Pi via ES and then turn the power off, I noticed that the power is still being supplied to the Pi (lights on the Pi are still on).
When the PCB SAFE SHUTDOWN is Off and I switch the power button off, it kills power to the Pi (lights on the Pi are off).
Could this still be a GPIO problem?

codecomp, 2019-01-07:
Ok so new case, and no problems. So something is not sending signal to the gpio pin. Not able to tell if it's the connector, the cable, the join to the board or something on the board. 

When i get a chance I'll borrow a multi meter to check further.

daboodamonk, 2019-01-07:
Thanks @codecomp 
I've ordered yet another case (this time from GeeekPi vs RetroFlag on amazon canada) and also another Pi 3B+.
Lets see if I noticed any difference.
I will post when I try this out.

codecomp, 2019-01-07:
My new case came from Geeekpi, the box was slightly different to the one I got from Pi hut that was broken.

Fozziefett, 2019-01-07:
Thanks! What script are you using with the new case?

CookiePLMonster, 2019-01-07:
For the record, my case also came from GeeekPi - no problems.

codecomp, 2019-01-07:
I'm currently using the custom one by @crcerror Though I'm not sure if his script will detect the newer cases as the script does seem to determine what to install based on case.

There is no difference between his script and the normal one as far as gpio pins and their calls as far as I recall, so if it's hardware either would have worked / not worked for me.

crcerror, 2019-01-08:
@codecomp The script will also work on the MegaPi as you already mentioned there is nothing altered from pin connection between the one from RetroFlag and mine

daboodamonk, 2019-01-09:
I reached out to Retroflag as well and this is what they had to say
**_I am so sorry about the problem that you are having with the case.
It quiet similar with this post: https://www.reddit.com/r/RetroPie/comments/9jtpap/safe_shutdown_still_not_working_on_replacement/
And we made some tests about this, it turns out that the GPIO was burned._**

Hope this problem goes away once I test it with a Pi.

Fozziefett, 2019-01-09:
In my case, my Nes Retroflag shuts down fine but the MegaPi case on the
same Pi does not with the Retroflag script. I will try another case and can
also try my spare Pi.

- Mike

codecomp, 2019-01-12:
@daboodamonk What happens with the non working pi in the working case? and the working pi in the non working case?

darty82, 2019-06-16:
I had this same issue but when looking at the python scripts I was sure it was a scripting issue not a hardware issue.

I tried an alternative script posted earlier in this thread ( not sure that it matters but I made sure that safe shutdown switch was on and the power switch was on too) and it worked perfectly.

The script was:

https://github.com/crcerror/retroflag-picase/

The read me on this GitHub shows the exact command ( which is identical to the official command but with the URL swapped out).

Ultima2876, 2020-03-25:
I had the same issue, but as @darty82 said, the modified crcerror script works.

https://github.com/crcerror/retroflag-picase/

oskich, 2020-05-03:
I have a MegaPi Case and a 3B, where I couldn't get the script to run. When I flipped the switch it only activated the LED blinking - No shutdown process initated...

I changed the shutdown-string in /opt/RetroFlag/SafeShutdown.py from:
**"os.system("sudo killall emulationstation && sleep 5s && sudo shutdown -h now"**
to
**"sudo shutdown -h now"**

Now the buttons work - does it matter that I don't kill the ES-process befor shutdown and reboot?

PeaceFrogger, 2020-06-03:
I just tried to install the script. The led shut off and the power and reset buttons did nothing. I figured out that I installed the script wrong. When typing in the website (wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/install.sh" | sudo bash) it was the symbol before sudo that messed me up. 
[image]. I found the double vertical dash is the symbol

Dragonlord944, 2020-12-25:
the trouble I just had getting to that symbol. numerous keyboards trying to get it. a royal PITA! thanks for pointing that out. Mines now working perfectly again

fab1o, 2022-10-17:
You actually don't need to type this pipe symbol `|` you can just do:
```sh
wget "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/install.sh
```
this will download the file, and then you do:
```sh
sudo ./install.sh
```