RetroFlag/retroflag-picase #32: Attract Mode

faustable, 2018-10-01:
I'm using a pi3 B+ and the Retroflag Superpi Case.The case and  script works perfect if I'm using emulation station but if I switch to Attract Mode the on/off and reset button do not work so I have to shut it down through the menu.  Is there a script that works with Attract Mode?

bhazard451, 2018-10-02:
I have the same issue. The power button blinks nonstop when I shut off the power switch. It does not send the command in Attract Mode.

faustable, 2018-10-03:
I ran sudo nano /opt/RetroFlag/SafeShutdown.py and changed wherever it said emulatationstation  to attract and that seemed to work. Both the power button and reset work and saves metadata.

closed 2018-10-03 by faustable

mel12n, 2019-07-11:
@faustable can you describe wich lines i have to edit in the script? i read it and it calls functions on the other script (Multiswitch) thanks!