RetroFlag/retroflag-picase #33: How to use with Raspbian?

Paradoxum, 2018-10-06:
How to use with Raspbian?

ChatFou, 2018-10-08:
I found a video on Youtube with instructions of adjustments to make it compatible with raspbian:
https://www.youtube.com/watch?v=kaEP6Nu7zvA
Hope it helps.

Paradoxum, 2018-10-16:
Thanks. Hopefully Retroflag will add a script for the buttons using Raspian/Lite.

keilmillerjr, 2018-11-26:
Follow the example for „RetroPie“. I have no idea why they call it that. Remove the bits about emulation station. I have no idea why its even there because the halt command will shutdown any running process. It’s redundant.

djayor, 2021-08-31:
> I found a video on Youtube with instructions of adjustments to make it compatible with raspbian:
> https://www.youtube.com/watch?v=kaEP6Nu7zvA
> Hope it helps.

I try this method with the NesPi 4 but didn't work.
The power button just cut the energy immediately, and reset instead of reboot command just rest the RPI.

Will be great if we had the support from RetroFlag for Raspbian OS.