RetroFlag/retroflag-picase #41: powerenPin? No documentation.

keilmillerjr, 2018-12-17:
```
powerPin = 3 
resetPin = 2 
ledPin = 14 
powerenPin = 4 
```

I have a megaPi case. Documentation is non existent in the script. I am assuming the following:

powerPin (3) is the sliding switch’s on state.
resetPin (2) is the physical button.
ledPin (14) is the output to the red led.

What is powerenPin (4)? I don’t get it.

CookiePLMonster, 2018-12-26:
Basing on my observations, pin 4 gets monitored by the case circuitry. When you set your case to use Safe Shutdown and then set it to high from a script, then circuitry monitors its state, and once it goes low it cuts power entirely.

It's basically a "it's safe to cut off power now" marker.

keilmillerjr, 2019-01-01:
I am not 100% sure on anything @CookiePLMonster. There is no documentation. I am only guessing my looking at the code. I wish @RetroFlag would comment on this.

CookiePLMonster, 2019-01-02:
So `powerenPin` seems to be only a marker for case's PCB on when it's safe to cut the power. You can verify it by sliding the power switch to OFF and then turning this pin off.

CookiePLMonster, 2019-01-02:
Confirmed `powerenPin` empirically. I slided the power switch to OFF and executed the following:

[image]

Pi did shut down (forcibly, not gracefully) after executing this.