RetroFlag/retroflag-picase #49: Please support Lakka，thx！

CkauNui, 2019-01-23:


cliffyb73, 2019-02-13:
Yeah I agree, please support Batocera

CkauNui, 2019-02-25:
This work!
 wget -O - "https://raw.githubusercontent.com/jucalize/retroflag-picase/master/batocera_install.sh" | bash

cliffyb73, 2019-02-26:
works like a charm! ty

nadenislamarre, 2019-06-20:
batocera 5.22 already includes these scripts.
batocera 5.23 will come with an out of the box support of the gpicase (ok, 1 file to rename to config.txt)

cliffyb73, 2019-06-20:
If 5.22 has it already built in, its not working. Just for good measure when you install the above script, all the settings reset and the bluetooth disbales.

228peace, 2019-07-05:
PLEASE try this 

For Batocera: 
wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/install_batocera.sh" | bash

from https://github.com/crcerror/retroflag-picase

> If 5.22 has it already built in, its not working. Just for good measure when you install the above script, all the settings reset and the bluetooth disables.

crcerror, 2019-09-28:
I revisted the script it will work for BATOCERA 5.24 and later -> This is newest and more advanced version
`wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/install_batocera.sh" | bash`

For versions **LOWER** than 5.24 try to use -> This version is quite old and I tested it only for BATOCERA 5.20 and 5.22
`wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/install_recalbox.sh" | bash`

RaM5eS, 2019-11-07:
How do I enter to terminal in Batocera 5.24? CTRL+ALT+F3 does not work, nor does F4.

I am using version Rapsberry Pi 3 5.24 Dev. (2019/10/22)

crcerror, 2019-11-07:
BATOCERA does not clear the framebufferfor Raspberry. So only chance to enter terminal is per SSH.
Maybe they add a **Quit to terminal** option later

RaM5eS, 2019-11-07:
Thanks friend, I finally did it through SSH as you said.

crcerror, 2019-11-07:
@RaM5eS Just as info, in the 2019/10/22-version is an error in BATOCERAs EmulationStation! This version can't save any metadata if there is a powerdown from outside :(

You can try newest version! 
https://batocera.org/upgrades/rpi3/stable/last/next/