RetroFlag/retroflag-picase #50: Works in Emulation Station but not Attract Mode

lghtanddrk, 2019-01-26:


lghtanddrk, 2019-01-26:
I am using Regalad Supreme Build V2 on a Pi 3 B+. The shutdown and reset work in Emulation Station but not if I reboot into Attract Mode. Any way to get it to work in Attract Mode?

keilmillerjr, 2019-01-26:
Shutdown and reboot are system commands that should work regardless of what app is open. Please post your SafeShutdown.py file.

keilmillerjr, 2019-01-26:
/opt/RetroFlag/SafeShutdown.py

Use scp to copy the file locally, or on the pi from terminal use nano and snap a picture.

I am not familiar with retropie. Tried it once and wasn’t happy with how slow it was. I’m using raspbian lite. However, the way this script works should yield similar results across different Linux distros on the pi.

mel12n, 2019-07-08:
I have the same cuestion, ¿any solution?