RetroFlag/retroflag-picase #55: Issues with build quality and safe shutdown for SUPERPI UCase

ellereeeee, 2019-03-11:
Hello,

I recently purchased a SUPERPI UCase from Amazon for my Raspberry 3B+ RetroPie.

**I have a couple issues with the build quality and script that runs the safe shutdown and reset. I'll try to concisely describe the issues and the troubleshooting steps I've taken.**

1) The screws included that assemble the case are too large and stress the plastic posts on the bottom half of the case. The plastic posts are visibly stressed and white after forcing the screws in. Here is a [photo](https://i.imgur.com/F5a8YHa.jpg). While I'm unhappy with this, the case does assemble.

2) Power down and reset do not work correctly with the safe shutdown PCB switch turned on.

I used the RetroPie image for 2/3 here: https://retropie.org.uk/download/

After assembling the case with my 3B+ and setting the PCB safe shutdown to on, I entered this script from your github repo: wget -O - "https://raw.githubusercontent.com/RetroFlag/retroflag-picase/master/install.sh" | sudo bash

**When the "safe shutdown" switch on the PCB is on:**
- The pi turns on when I turn on the power switch. 
- The pi does not turn off when I turn off the power switch. 
- The pi runs a shutdown script when I press reset, but Emulationstation is not booted up again. The pi is powered on (indicated by the red LED) but nothing is sent to the monitor via HDMI. 

**When the "safe shutdown" switch on the PCB is off:**
- The pi turns on when I turn on the power switch. 
- The pi turns off immediately (no code displayed) when I turn off the switch. 
- Emulatorstation resets immediately (no code displayed) when use the reset switch.

I've tried using the script by crerror but I keep getting a 404 response in the RetroPie terminal.

One thing I wonder that might be an issue is that the wires that connect the case to my Raspberry Pi are colored differently than the instruction manual and what I've seen in online tutorials. Here is a [photo](https://i.imgur.com/aN9tFk9.jpg) of my wires.

To be frank, I've been unhappy with the instructions included with the case, the build quality, and the script. I would appreciate it if you could help me troubleshoot the issue.

If I can just get the power and reset buttons to work correctly then I'll be satisfied.

mrdink, 2019-03-25:
I'm having the same issue with mine.

yoyojacky, 2019-03-27:
I have this too, but i have already fixed this problem, I've follow this 3 step to set it up.
1.  press F4  when it finished booting up.
2.  Connect My Pi to internnet via "sudo raspi-config" tool and at the same time i enabled ssh so that i can operate it remotely.
3. Execute the screen as comment before. 
wget -O - “ https://raw.githubusercontent.com/RetroFlag/retroflag-picase/master/install.sh “| sudo bash
4. and 3 seconds it reboot automatically...
5. upload my Nintendo Game Rom files to /home/pi/RetroPie/roms/nes/ 
6. reboot it ,and seems like everything is working properly ..

ellereeeee, 2019-04-06:
@yoyojacky Thank you for helping.

I got a response from Retroflag via email and it turns out I was sent faulty hardware. The wires in the molex connector that connects the case to the PCB were wired incorrectly. My replacement works fine.

closed 2019-04-06 by ellereeeee