RetroFlag/retroflag-picase #57: Shutdown & Restart Without Dropping to Shell

idemiq, 2019-04-16:
At present, invoking SafeShutdown by using the Picase buttons closes ES and presents the Command Line generally - and for the duration of the Wait.

Is it possible - or would it be possible - to either "quiet" this by passing it to tty3 for example, or obfuscate it by invoking a graphic/message?

idemiq, 2019-05-08:
For the time-being I added the `-13` flag to each KILLALL, and `&>/dev/null` to each SLEEP.

In case anyoneelse wanted to quiet SafeShutdown.py

d-rez, 2019-09-20:
Changing `console=tty1` to tty3 would probably work for both startup and shutdown 

See https://github.com/RetroPie/RetroPie-Setup/wiki/FAQ#how-do-i-hide-the-boot-text