RetroFlag/retroflag-picase #60: Reset Button always reboots the system instead of jumping to ES

xadox-1st, 2019-05-20:
Using this on a SuperPi Case with a Raspberry Pi 3 B+.

The readme describes the following:
_If you press restart if emulator is currently running, then you will be kicked back to ES main menu.
If you press restart in ES main screen, ES will be restartet (no reboot!), good for quick saving metadata or internal saves._

Is the readme outdated and the reset button always makes a complete reset? Or does the script not behave correctly for me?

xadox-1st, 2019-05-22:
Installed the RetroPie from scratch and use the **Multi Switch Shutdown** Script by @crcerror now.

vguttmann, 2020-04-09:
Have teh same Problem too. Don't they use the @crcrerror script here also? Readme.md suggests that

vguttmann, 2020-04-22:
I don't think reinstalling is necessary. Updating RetroPie also removes it. And would it be possible to add this script as a package for RetroPie?