RetroFlag/retroflag-picase #64: Unnecessary configuration in GPi Case Patch?

franklesniak, 2019-06-23:
Hello,

This comment does not pertain to the safe shutdown scripts, but rather the GPi Case patch located at download.retroflag.com

I have noticed a few configurations that appear unnecessary and one possible problem that I wanted to highlight:

- In the ZIP file, the config.txt file format uses Windows-style line breaks instead of UNIX-style. Since the config.txt file included with Raspbian uses UNIX-style, I believe this should be corrected.

- The config.txt file instructs the bootloader to load the "zImage" kernel, but it is not included by default, as far as I can tell, in the RetroPie 4.4 image file for the Raspberry Pi Zero. Should this item be removed? Or can you clarify why it is needed?

- The config file changes the amount of GPU memory on systems with 1024 MB RAM from the default 256 MB to 512 MB (gpu_mem_1024=512). This is not necessary on a Raspberry Pi Zero because the RPi Zero only has 512 MB of RAM; therefore, this configuration would be ignored, right? If I am correct, this line item should be left at its default.

- The lines that set overscan values to 0 (overscan_left=0, overscan_right=0, overscan_top=0, and overscan_bottom=0) are setting default values and therefore should not be necessary. I believe they can be removed without consequence.

- The line setting the HDMI CEC name (cec_osd_name=recalbox) is not necessary because the HDMI signal cannot be sent outside of the closed system and would never be seen. I recommend removing this line.

- Why is the config.txt file rotating the display by 90 degrees (display_rotate=1)? Is this necessary?

franklesniak, 2019-06-28:
Related, please see: https://www.reddit.com/r/retroflag_gpi/comments/c2q7ho/help_black_screen_after_updating_retropie/es8b8z6?utm_source=share&utm_medium=web2x

d-rez, 2019-09-20:
It's leftover default config values, they're not hurting anyone