RetroFlag/retroflag-picase #70: No issue, just need a helping hand.

testbughub, 2019-09-17:
Hi!

I recently got my GPi case, and all works well.
However, I would like to display an image as the device is shutting down.
I've tried adding an fbi command before the rest of the shutdown script, but this doesnt seem to work, as it just shuts down right away.

Could anyone help me with this?

d-rez, 2019-09-20:
This is totally irrelevant to this repo. But I've done this already, however I haven't shared anything.

Basic principle is:

* pngview
* systemctl
* systemctl service unit file that only executes when runlevel indicates that OS is shutting down

Good luck. I might share mine at some point but google should get you sorted

d-rez, 2019-09-20:
Also duplicate of #34

testbughub, 2019-09-20:
Yeah, I was afraid I might be in the wrong place. Gonna try at reddit.
Thanks for letting me know about the dupe. Thought I had looked enough beforehand.

closed 2019-09-20 by testbughub