RetroFlag/retroflag-picase #73: Wont turn on or shut down

smacknuts, 2019-11-25:
When the Safe Shutdown switch is in the ON position, the power won't turn on or off.  The script seemed to download okay, I didn't get any errors.  I am sorry if I sound like a noob.

crchauffe, 2019-11-30:
I was having the same issue. The script installed just fine but none of the switches were working. I switched the safe shutdown switch to OFF and it started working just fine?  (which doesn't make any sense to me)  I guess try playing around with the safe shutdown switch and see if that helps.

Dannyguk, 2019-12-02:
I"m having a similar issue, but it doesn't seem to work with the switch OFF or ON :/

smacknuts, 2019-12-02:
I found a solution to this issue.  Basically I scrapped this script and found a custom script somewhere else.  I used the one here...

https://discourse.osmc.tv/t/howto-install-safe-shutdown-script-for-retroflag-cases/78572

Zanmato27, 2019-12-27:
> I found a solution to this issue. Basically I scrapped this script and found a custom script somewhere else. I used the one here...
> 
> https://discourse.osmc.tv/t/howto-install-safe-shutdown-script-for-retroflag-cases/78572

Thank you, this worked unlike this obsoleted script that isn't working on newer RetroPie builds.