RetroFlag/retroflag-picase #82: GPi Safe shutdown script doesn’t trigger

runningdemon, 2020-05-11:
In the Retroflag GPi I have gone and installed and have everything working with the exception of the safe shutdown script.  

I have gone and installed it, but when I flip the on/off on the GPi, it just sits there in the off position and never triggers the script.

Any ideas on how to fix it?

oskich, 2020-05-11:
It didn't work for me either - had to edit the shutdown command to get it working(Retropie 4.6):

**/opt//RetroFlag\/SafeShutdown.py**

`os.system("sudo killall emulationstation && sleep 5s && sudo shutdown -h now"`

I removed this, and then it started working:

> killall emulationstation && sleep 5s &&

ds17f, 2020-05-11:
A more robust solution can be found in my PR: #84

Kevin-CodeCrew, 2020-06-28:
None of these solutions worked for me. When starts booting OS led turns off and buttons do nothing even with mods above. thanks

3wcircus, 2020-06-28:
None of these solutions worked for me. When starts booting OS led turns off and buttons do nothing even with mods above. thanks