RetroFlag/retroflag-picase #87: Scripts not working anymore - RetroPie

harukasan79, 2020-05-13:
Hello, I have a SuperPi case and I've always happily used the safe shutdown/reset scripts. I have installed last build of RetroPie 4.6 on a new SD card, but even if I have installed the scripts as usual, without any problem, they're not working anymore. I press the Reset button, or switch the Power button, and it does nothing. Could it be a problem with new Raspbian Buster? Thank you

Kevin-CodeCrew, 2020-06-28:
I am seeing the same issue. Ensured jumper/switch set to safe and scripts wget and install without error, but LED shuts off during boot and neither button works. thanks

Lazza, 2021-07-18:
I had the same issue. It turns out I forgot this first step:

> Turn switch "SAFE SHUTDOWN" to ON.

If you do not do this, the script installs another Python program that does not work with retroflag-picase (nespi+, superpi, megapi,nespi4 case).