RetroFlag/retroflag-picase #90: NESPi 4 Case Shutdown Scripts Not Working Correctly

jasonkonen, 2020-08-15:
When I hit the power button to shut down the system, it goes through the shutdown procedure, but it then reboots immediately.

Olliek5, 2020-08-16:
Same problems on latest retropie and and light pi is image using NESPi 4

Shaddowbad, 2020-08-17:
Hello I have the NESPI4 case with a Raspberry Pi 4 8GB. I installed the Safe Shutdown Script. If I switch off the case with the power button, it does not go off. It started all over again. The mistake needs to be heard around.

crcerror, 2020-08-18:
This one will work!
https://github.com/crcerror/retroflag-picase

Install for RetroPie
`wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/install.sh" | sudo bash`

Uninstall for RetroPie
`wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/uninstall_all.sh" | sudo bash`

noracegres, 2020-08-19:
> This one will work!
> https://github.com/crcerror/retroflag-picase
>
> Install for RetroPie
> `wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/install.sh" | sudo bash` [...]

Hi crcerror, your patch works to shutdown the pi, but the fan his still running after shutdown, any idea ? 

Thanks

crcerror, 2020-08-19:
@noracegres Do you have an Pi4? You can try to add
`dtoverlay=gpio-poweroff,gpiopin=4,active_low=1,input=1` to /boot/config.txt

noracegres, 2020-08-19:
> dtoverlay=gpio-poweroff,gpiopin=4,active_low=1,input=1

My bad, I forgot to mention that it was for the new nespi4 case.

Thanks, I'll try your config and come back.

roccop20, 2020-08-31:
I have a similar problem, when I shutdown the pi via the button, it reboots and after booting (on the black screen with the cursor) it shuts down. Do you have any idea of what happening? the reset button works fine

jeffsawatzky, 2020-08-31:
@jasonkonen @Olliek5 I just tried with the latest script that was updated after this issue was created. It seems to have fixed my issue, perhaps it will fix yours too.

Not sure if you need to uninstall the old script first (I did) before you can install the latest one though.

OfficialPhilcomm, 2020-10-07:
> Hi crcerror, your patch works to shutdown the pi, but the fan his still running after shutdown, any idea ?
>
> Thanks

In `/opt/RetroFlag/SafeShutdown.py`, replace the `poweroff()` method with
```
def poweroff():
	while True:
		#self.assertEqual(GPIO.input(powerPin), GPIO.LOW)
		GPIO.wait_for_edge(powerPin, GPIO.FALLING)
		GPIO.output(powerenPin, GPIO.LOW)
		os.system("sudo killall emulationstation")
		os.system("sudo killall emulationstatio") #RetroPie 4.6
		os.system("sudo sleep 5s")
		os.system("sudo shutdown -r now")
```

That does the trick for me.
I wanted to create a pull request for that fix, but the admins don't give the permission for open PRs

crcerror, 2020-10-07:
You'll instantly lose power with this if you use the NESPi power button. If you're happy with this then go on.

OfficialPhilcomm, 2020-10-07:
True, so nevermind, that is not the solution

BenMcLean, 2020-10-16:
Yikes. Retroflag really needs to update this.

FRM-CCA, 2020-10-24:
Yes same problem on nespi 4 with ssd, batocera 5.27, button reset force reboot system but reboot always, button power shutdown system but reboot

tested with /boot/config.txt
dtoverlay=gpio-poweroff,gpiopin=4,active_low=1,input=1

tested with batocera.conf
system.power.switch=RETROFLAG

crcerror, 2020-10-24:
User error - please uninstall retroflag script first or remove custom.sh file

FRM-CCA, 2020-10-25:
Thank you remove custom.sh, uninstall and reinstall, OK :)

skulltre, 2020-11-04:
Hi guys, I'm new here. Been trying all the suggestions here but i couldn't get my Nespi 4 to shut down properly. My pi 4 8gb is already on the latest EEPROM and booting to retropie on SSD. 

Firstly, I am told that the front panel LED lights will not function as long as the script is not running correctly if safe shutdown is set to ON, is this true? On off switch it lights up a constant red.

I first tried the script by retroflag, i can see the script is installed via ps aux | grep py . however the script wouldn't work, nor the LED lights. I have since tried uninstall and remove the SafeShutdown.py and the entry in /etc/rc.local and proceeded to reinstall - still no luck. 

any ideas?

SolidHyunkel, 2020-11-18:
Hey! The scripts didn't work for me too... My setup is nespi 4 case, raspberry pi 4 8gb running the latest retropie. When are they going to patch this? It doesn't seem to be too hard.... Good luck to all!

skulltre, 2020-11-18:
I recently managed to replace the control board of the Nespi4 case and the Retroflag script works. But the front panel LED lights would not work on my pi4 8gb. If the safeshutdown switch is set to off there will be LED lights on at front panel. Doing a clean uninstall and using crcerror's shutdown script never worked for my pi4 8gb either. I am now waiting for a pi4 4gb board to arrive to test.

bezaban, 2020-12-12:
I've got a nespi4 case and a rpi4 8gb.  I'm having similar issues. I've tried both crcerror and the official scripts.

Did a little bit of debugging and it seems it doesn't pick up the event or is listening on the wrong channel - as GPIO.wait_for_edge() never triggers @ https://github.com/RetroFlag/retroflag-picase/blob/master/SafeShutdown.py#L27

rtarroyo, 2020-12-13:
I'm having a similar issue, using a RP4 4gb with Retropie, when I press the power off button it goes through the motions of shutting down an then reboots before actually shutting down.

Also, the fact that it shows the terminal before shutting off makes it ugly when you are going for the console experience, I'm trying some changes but so far nothing

rtarroyo, 2020-12-14:
> This one will work!
> https://github.com/crcerror/retroflag-picase
>
> Install for RetroPie
> `wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/install.sh" | sudo bash` [...]

This worked! Wow, thanks! But apparently it generates a conflict with emulationstation, that if you shutdown through it everything will crash

bezaban, 2021-01-12:
I think the issue here (rebooting with -r) is worth fixing.  I first suspected my issue was related, but after digging in it's clearly something entirely different.  I have tested my nespi4 case with two different RPI 4s, tried a multitude of GPIO libraries, got in the middle with jumper wires and have excluded mostly everything than a faulty PCB.  

I'm still a little confused on what role the "powerenpin" plays in this, but it seems clear that the power switch never produces a falling edge, yet I can short GPIO3 to ground and it works as expected.   I've also realized that the board seems to power up on 5v input rather than shorting GPIO3. 

Maybe a little malplaced here, but posting for posterity.

thehawkyo, 2021-02-09:
question, I installed RetroPie on an SD and I am in the command prompt attempting to use the command 

wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/install.sh" | sudo bash

but when typing it out, it shows as this for some reason ( " will show up as @, and | as ~), I have no idea what is going on

wget -O - @https://raw.githubusercontent.com/crcerror/retroflag-picase/master/install.sh@ ~ sudo bash

and of course spits out errors:
scheme missing
http://sudo/ resolving sudo (sudo): failed
wget: unable to resole host
resolving bash (bash): failed

mickelsn, 2021-02-14:
> question, I installed RetroPie on an SD and I am in the command prompt attempting to use the command
>
> wget -O - "https://raw.githubusercontent.com/crcerror/retroflag-picase/master/install.sh" | sudo [...]

You need to go into the raspi-config menu and set your localization options to the correct keyboard. I believe the Raspberry Pi 4 ships configured default for Great Britain; I had to change to US (locale and keyboard) and this solved my problem.

Lazyboi-93, 2021-04-05:
Amazing, thanks everyone. This was really helpful. Works as expected.

jessieknifgt, 2021-04-25:
I have the same problem but I'm running Ubuntu 20.10 OS. I'm assuming this script is for the retro-pi OS. is there one i can use just for ubuntu?

m0uh85, 2021-04-30:
I would like to use this case (NESPI 4) with Raspbian - is there any chance to get a working shudown script for Raspbian?

Marsian9, 2021-07-06:
question when i turn off the power switch on nespi 4, it run the safe shutdown scripts but few sec the led comes back on and keep flashing every 5 sec, my led on the controller on and off every few sec, and the fan is on the whole time.
i have tried uninstall the safe shutdown scripts and reinstall it still same thing happen. 
can anyone help?

zero4281, 2021-07-06:
> question when i turn off the power switch on nespi 4, it run the safe shutdown scripts but few sec the led comes back on and keep flashing every 5 sec, my led on the controller on and off every few [...]

Try updating the firmware for the Pi and the case.

Marsian9, 2021-07-07:
i did both firmware updated already

> Try updating the firmware for the Pi and the case.

i already updated both firmware

deatheffect, 2022-02-09:
> In `/opt/RetroFlag/SafeShutdown.py`, replace the `poweroff()` method with
>
> ```
> def poweroff():
> 	while True:
> 		#self.assertEqual(GPIO.input(powerPin), GPIO.LOW)
> 		GPIO.wait_for_edge(powerPin, [...]

^^^^^^ using the stock retroflag script and commenting out / replacing the poweroff with this worked for me, thankyou dude!