RetroFlag/retroflag-picase #92: pi 4 the nespi4 case with Recalbox 7.0.1 does not turn off the pi power

thanosazlin, 2020-10-25:
i have a new pi 4 just week old with your nespi4 case, i putty in and ran the cmd for recalbox. it updated and rebooted, but when i push the button to turn it off (th  power button depressed) it does shutdown recalbox but the fan continues to run on the pi4 and the pi4 doesn't cut power off :(.  so i have to pull the power plug and plug it back in very annoying :), but if you need any config files or for me to try anything pls let me know.

thanosazlin, 2020-10-27:
hmm seems to be working suddenly which is very weird i haven't changed anything just thought i would let you know

SeanDKendle, 2020-12-30:
I'm having this exact issue and haven't found anyone else talking about it! It's very frustrating. It seems the times it actually does power off the Pi and fan it tries to reboot (seemingly) and then finally shuts off. This can't be good for the RPi if it's actually powering down mid-boot.

I'm using an SSD (obvs, that the point of this case).

Richard-RTA, 2022-05-29:
I also have this exact issue.  I'm using the superpi4 case, but using the retropie version of the scripts, and I have the fan installed too.
Three items to be clear on.
The power button on the case, the led on the case, and the led on the raspberry pi board (which you can _just_ see shining through the case near the SD card slot).

So....
power button ON, and then the led on the case and the board come on.  Everything booting up. Fine.
switch the power button OFF, and you can see the kill emulation station in effect, and the shutdown happening, and the case led switches off...but the fan continues, and the board led is still lit.

Never used to happen on my pi3+ - it all worked perfectly.

I'll keep experimenting, and will post up if I discover the answer.