RetroFlag/retroflag-picase #93: NesPi4 : Safe Shutdown script Not working!

msx2p, 2020-11-07:
Nespi4 case
Safe shutdown script not working!
I installed script correctly, 
I turned on switch.
I pressed the power button but it didn't turn off.
I pressed the reset button but did not restart.
After I disconnected the usb1(ssd) cable it worked fine.
When I connected the cable again, it did not work.
I use retropi 4.7.1 version, and the eeprom version is September 3, 2020.
I don't use ssd.
Is there no way to solve it?

msx2p, 2020-11-17:
I solved this problem.
I updated the firmware of the usb module adapter.
The shutdown script worked perfectly.
I couldn't see the warning statement that there was a problem with the existing firmware anywhere in the manual and on the homepage of the product I purchased.
Why doesn't the manufacturer notify customers of a product defect?

closed 2020-11-17 by msx2p

magostinelli, 2020-11-17:
Can you post the intruction to update the firmware of usb module? Thanks.

msx2p, 2020-11-17:
> Can you post the intruction to update the firmware of usb module? Thanks.

Visit following link.
http://retroflag.com/nespi-4-case.html
You can find "firmware download"
Download the file.
Installation instructions are included in the file.

Note for the second installation item:
Connect the nespi4 case's ssd usb cable to the Windows PC and update the firmware.