RetroFlag/retroflag-picase #98: NesPi 4 Case EEPROM Issue

TechXero, 2020-12-13:
Hello,

I dunno if it's right place to ask but I have updated to latest beta eeprom which dates to December 11th.. Now the HDD inside my NesPi 4 Case will no longer be detected.

I am running Batocera booting from SD card and ROMs on 2TB HDD. Batocera boots just fine but will no longer see my ROMs HDD. I checked in another OS I have on another SD card same issue drive no longer there. I have added the usb.quircks line to cmdline.txt to no avail. I am not booting from HDD. I just want HDD to be detected.

Tested the following :

Raspberry Pi OS
Twister OS
Batocera
RetroPie

Same result on all. Only EEPROM that works is Stable September 03 build anything beyond that renders my HDD useless...

Note : I updated controller firmware from your site. Not fixed

2play, 2020-12-16:
There are at least 4 people i know having issues with the JMicron controller (?) on nespi 4 case not booting  USB drives
as @TechXero some on older pi4 fw (sept 20 or older) can boot by disabling UASP by blacklisting it with quirks line mentioned above.

TechXero, 2020-12-16:
Yep here's what happens after firmware upgrade :

[https://i.imgur.com/rYYmYD3.png](url)