RetroFlag/retroflag-picase #101: Pi 4 hangs up on soft shutdown and reboot.

Brayton898, 2020-12-24:
Hi, 
     I have a PI 4 with retropie version 4.7.1 that hangs up on soft shutdown and reboot using the the softshutdown feature. the line it hangs up on says "stopping restore/save the current clock". I have to pull the power cord in order to reboot the pi.

Zizzax, 2021-01-17:
Hey,
I have the same problem on my Pi 4 4GB with a NESPi4CASE running retropie V 4.7.1 
Hopefully Retro Flag will release a new updated Safe Shutdown script that works on retropie V 4.7.1 soon.

HellesVollbier, 2021-02-06:
Hi,
I had the problem after overclocking my PI.
I`ve just reinstalled the script "https://github.com/crcerror/retroflag-picase" and now it works again.

Zizzax, 2021-02-06:
Cool,
Thanks for the info I will try reloading the crcerror's version of the script.

X52p, 2021-04-12:
I have the same problem, but I didn't overclock and also reinstall didn't fix it.

HellesVollbier, 2021-04-15:
@X52p Did you use the original Retropie script or the crcerror version?

X52p, 2021-04-15:
I tried both, but the problem happens with both.
It also seems that it doesn't happen all the time but only most of the time. Sometimes it shuts down just fine.

ALLRiPPED, 2021-04-20:
having same issue with both scripts, yes I am over clocked and even not OC'd and fresh install it hangs on reboots and shutdowns

lyat97, 2021-04-25:
This problem is caused by the firmware of the SATA to USB adapter. The firmware can be found in retroflag. The UAS version can be used for SD card boot / no UAS version is suitable for USB boot. If you boot from the SD card and use the no UAS version of the SATA drive firmware, the shutdown and restart will be frozen

ALLRiPPED, 2021-04-25:
okay I finally got mine working with a simple install of the drivers of crcerror's scripts, all I did was flash a fresh copy of retropie to the sdcard, then after boot and setup you install crcerror's https://github.com/crcerror/retroflag-picase after that reboot, DO NOT install the one from retroflag, AT ALL, it will make it hang up, that was my problem at first I would have one installed on top of the other and neither would work.

Un0Du0, 2022-02-01:
> This problem is caused by the firmware of the SATA to USB adapter. The firmware can be found in retroflag. The UAS version can be used for SD card boot / no UAS version is suitable for USB boot. If [...]

Installing the UAS firmware from here worked: https://support.retroflag.com/Products/NESPi_4_CASE/How_to_boot_from_SSD.zip

Zizzax, 2022-02-02:
> I was having this error with both the crcerrror shutdown and the retro-flag picase shutdown scripts. Apparently the Raspberry Pi 4 boards ccan have a cofnlict shutting down the SCSI driver. I got around this by modifying /opt/RetroFlag/SafeShutdown.py to manually kill usb2 before shutdown. This worked for me
>
> #waits for user to hold button up to 1 second before issuing poweroff command
> def poweroff():
>         while True:
>                 #self.assertEqual(GPIO.input(powerPin), GPIO.LOW)
>                 GPIO.wait_for_edge(powerPin, GPIO.FALLING)
>                 os.system("sudo killall emulationstation")
>                 os.system("sudo killall emulationstatio") #RetroPie 4.6
>
>                 os.system("sudo chmod 777 \/sys\/bus\/usb\/drivers\/usb\/unbind") #allow for all to unbind
>                 os.system("sudo echo \"usb2\" > \/sys\/bus\/usb\/drivers\/usb\/unbind") #turn off usb hub
>
>                 os.system("sudo sleep 5s")
>                 os.system("sudo shutdown -r now")

Thanks for the info M8Good job I will have to take my NESPI out of storage and modify the script.Hopefully I will have luck finding my NESPI in my storage unit it's packed to the celling.I guess I will have to play real life Tetris soon.I will let you know if the code modification works for me.
Cheers

Zizzax, 2022-02-02:
Thanks Shaun and Michael for the valuable info.