RetroFlag/retroflag-picase #109: Hardware > PI Does not shut down, only restarts when case is put together.

alexhofer, 2021-01-19:
I'm not sure where else to raise this issue since I can't find a support email, but maybe it will help someone else. 

I was having an issue where the safe shutdown was only ever restarting the PI only when the case was put together whether or not the power switch was on or off, if this happens to you it's probably not the issues with the shutdown script but the little circuit board that takes in power for the case. Mine had been touching the PI itself on the power contactors and causing a short circuit which would constantly keep it powered. Just throw some electrical tape on that little board and it should be fine after that.

ChromeFox, 2021-01-25:
Interesting observation. Do you mind providing a photo of where exactly you believe it short circuits or comes in contact with? I might be having the same issue here.

alexhofer, 2021-01-27:
> Interesting observation. Do you mind providing a photo of where exactly you believe it short circuits or comes in contact with? I might be having the same issue here.

Sorry for the wait, here are some images of the issue I was getting. Though I am not sure if maybe my PI board itself was a little warped or if the solder joints were just a little tall. 

[image]
[image]
[image]

ChromeFox, 2021-01-27:
Hey, I appreciate the photos. I think you have a different version of my case. I have the Nes Pi 4 Case and my case doesn't have that part. Interesting nonetheless.

I realized enabling the default fan in raspi configuration will interfere with one of the GPIO pins used for the circuit board of the POWER / RESET buttons.  That's what fixed it for me and disconnecting stuff and slowly connecting it together helped me recognize that my pi couldn't receive the "shutoff" signal.

alexhofer, 2021-01-27:
No problem! Ah mine is the 3b+ version, so it must work completely differently. I'm glad you got your issue figured out though! Hopefully this issue and your comment can help someone else!

ChromeFox, 2021-01-27:
Thanks and likewise, yours definitely helped me look into the right direction