RetroFlag/retroflag-picase #111: Shutdownscript does not work at all

schniedelcastro, 2021-03-06:
Dear all,

I have recently bought a nespi 4 case. I have setted my pi up with the latest version of retropie and I have also installed the shutdownsript in the way as it is described here.

Unfortunately, the power button works only for switching on. Afterwards I can click on it and nothing happens anymore. The LED stayed OFF during the whole time and the reset button does not work either.

I have added the following lines to ```/boot/config.txt```.
```ruby
dtoverlay=gpio-shutdown,gpio_pin=3
enable_uart=1
```

This enables the power switch an the LED. Unfortunately after shutting down, the fan stays active, which is very annoying. The reset button does not work still.

Concerning the fan I have also tried the following:
```ruby
dtoverlay=gpio-poweroff,gpiopin=4,active_low=1,input=1
```

Unfortunately, this does not help either. With adding this line, the system crashes during the shutdown.

My setup is as follows:
RPi4 8GB
Retropie v4.7.1

I hope you can help me, if you need further information, please feel free to ask :)

Best
Simon

Dirtywaterbrewing, 2021-03-21:
I too am experiencing the same issues ... hopefully someone has a patch ?

eborchardt, 2021-04-03:
I just installed it on an RPi4 4GB and it works fine for me. Are you certain you ran the correct installer? The installer is not the first option listed in the readme, that one is specific to the GPI case.

schniedelcastro, 2021-04-03:
Dear @eborchardt,
Thanks for the hint. I have recognized, that there are two scripts, after setting up the retroflag GPI case. I will give it a try.
Do you know, how to deinstall the script? Or is this not necessary.

Best
Simon

eborchardt, 2021-04-04:
Looking at the two scripts, it probably isn't necessary to uninstall the previous one. I would just try it.

schniedelcastro, 2021-04-05:
Dear @eborchardt,

you have pointed me in the right direction. As usually the problem is sitting in front of the computer ;) I have installed the script for the case and everything is woking just fine. Thanks. Ticket can be closed.

Best
Simon

closed 2021-04-05 by schniedelcastro