RetroFlag/retroflag-picase #115: Safe Shutdowwn Script does not turn off the fan in Recalbox 7.2 (NESPI 4 Case)

lullamolusco, 2021-04-28:
I have a Nespi 4 case and last week I switch from Retropie to Recalbox 7.2 and after installing the Recalbox safe shutdown the script I noticed that reset works fine but the power button shut down the system but not shut down the fan. 

I have Pi4 4gb an I am using a 64gb SD card. Did I execute the following command to instal the script:
wget -O - "https://raw.githubusercontent.com/RetroFlag/retroflag-picase/master/recalbox_install_gpi.sh" | bash

Could you please help me fixing this issue ?

TheDaveAbides, 2021-04-29:
I don't believe so. That looks like a game boy style case.  Here's instructions for the nespi4.

Example for RetroPie:

Make sure internet connected.
Make sure keyboard connected.
Press F4 enter terminal.
In the terminal, type the one-line command below(Case sensitive):
wget -O - "https://raw.githubusercontent.com/RetroFlag/retroflag-picase/master/install.sh" | sudo bash

lyat97, 2021-05-15:
> I have a Nespi 4 case and last week I switch from Retropie to Recalbox 7.2 and after installing the Recalbox safe shutdown the script I noticed that reset works fine but the power button shut down [...]

Note that you have installed GPI scripts. Please distinguish different scripts for different systems and hosts