RetroFlag/retroflag-picase #117: Unable to install safe shutdown script on Recalbox 7.2.1

Samington33, 2021-05-12:
Hi everyone,

I've just purchased and assembled the Nespi4 case. However, I unfortunately don't seem to be able to install the "Safe Shutdown" script, running Recalbox. In Terminal, after logging in as "Root", I executed the following command:

wget-o - "https://raw.githubusercontent.com/RetroFlag/retroflag-picase/master/recalbox_install.sh" | bash

I have included a screenshot of the resulting output, as I don't want to make a mistake when typing it out. 

The power and reset buttons do seem to be functional, and I do have a red LED. However, as I understand it, without the script, these buttons will cause the pi to shutdown incorrectly. Is this correct?

Hoping someone can advise. 

Any help appreciated. 

[image]

zero4281, 2021-05-12:
If you can't figure out what's going on with Recalbox then maybe try RetroPi.  I've always had a really good experiences with RetroPi.
https://retropie.org.uk/

Samington33, 2021-05-13:
> If you can't figure out what's going on with Recalbox then maybe try RetroPi.  I've always had a really good experience with RetroPi.
> 
> https://retropie.org.uk/

Thanks for the reply. This could turn out to be the best option, although I was enjoying Recalbox. 

Just wondering ... how can I tell whether or not the script is installed? For example, does the red LED still function without it? Does the power button still turn the system off and on; does the "Reset" switch still reset it, all be it unsafely? Basically I'm trying to ascertain whether Recalbox might have included the script in the newer builds, hence the seeming failure to install through Terminal. 

I should mention that I am registered as sight impaired, so apologies in advance if I've missed anything obvious. I can't read the manual; I put the case together with guidance from YouTube videos. 

Thanks again for any assistance.

Samington33, 2021-05-14:
So I've attempted to get this working under RetroPie. It seemed to install the script, then rebooted. 

When I attempt to power off or reset the system with the front buttons, RetroPie apparently tries to shut down. However, it gets stuck on "Stopping Restore / Save the CurrentClock". Any ideas?

As ever, any assistance appreciated.

lyat97, 2021-05-15:
> So I've attempted to get this working under RetroPie. It seemed to install the script, then rebooted.
>
> When I attempt to power off or reset the system with the front buttons, RetroPie apparently [...]

This problem is caused by the firmware version of USB tosata that supports SSD boot. If you only use SD card to boot, you can go to the official website to find the firmware upgrade software of SATA drive and use UAS firmware,

Samington33, 2021-05-15:
That worked great, thanks @lyat97 for the info. The scripts are now working perfectly on both RetroPie and Recalbox; I have a micro SD card for each of them. Really happy with my new console☺️.

closed 2022-05-23 by Samington33