RetroFlag/retroflag-picase #119: pin gpio3 already requested by leds

strangerting, 2021-05-28:
this seems similar to #2856

any ideas as to why this is happening on a pi4?  shutdown doesnt happen gracefully:-

[    7.868329] pinctrl-bcm2835 fe200000.gpio: pin gpio3 already requested by leds; cannot claim for soc:shutdown_button
[    7.868354] pinctrl-bcm2835 fe200000.gpio: pin-3 (soc:shutdown_button) status -22
[    7.868375] pinctrl-bcm2835 fe200000.gpio: could not request pin 3 (gpio3) from group gpio3  on device pinctrl-bcm2711
[    7.868397] gpio-keys soc:shutdown_button: Error applying setting, reverse things back
[    7.868435] gpio-keys: probe of soc:shutdown_button failed with error -22

zero4281, 2021-05-28:
I've noticed that there are a lot of issues on shutdown when the firmware for the Pi isn't up to date.

CarlOrQ, 2024-09-19:
被占用了检查是否占用它的引脚