RetroFlag/retroflag-picase #120: A physical pull up resistor is fitted on this channel!

alexsalex, 2021-06-10:
/opt/RetroFlag/SafeShutdown.py:15: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(powerPin, GPIO.IN, pull_up_down=GPIO.PUD_UP)
/opt/RetroFlag/SafeShutdown.py:16: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(resetPin, GPIO.IN, pull_up_down=GPIO.PUD_UP)
/opt/RetroFlag/SafeShutdown.py:17: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(ledPin, GPIO.OUT)

alexsalex, 2021-06-27:
Raspberry Pi 4
Fresh install of raspberry os, latest updates.

enazevedo, 2021-08-07:
Same here. 

Different from @alexsalex , I'm using VMAN custom RetroPie, so I initially thought that could be some kind of incompatibility. The message always shows up.

From tests, if the "SSD Sata Adaptor" into the USB (internal to the NESCase)is connected to the RaspPi (Pi4), the scripts doesn't work (nor Power nor Reboot). If Sata Adaptor is not connected, these message still shows up but the script/buttons works flawless.

cabaalexander, 2023-11-21:
this still persist

fab1o, 2023-12-28:
I also get this error on my superpi case. The case does not shuts down.  How to fix this Retroflag?

fab1o, 2023-12-28:
I also get this error on my superpi case (raspberry 3b+) using the latest update. The case does not shutdown.  It used to work fine but after this latestupdate, now it doesn't work anymore