RetroFlag/retroflag-picase #122: LED not working on Nespi4 case with safe shutdown switch to ON

kmarien, 2021-07-14:
I recently purchased the Nespi4 Case and a SSD for my raspberry pi 4B 4GB

With the switch to OFF and no script installed, the power button cut power and reset button resets the pi. The LED is ON.

But after I installed the script and turned the switch to ON, the buttons are working fine but the LED stays OFF.

I have this problem on Raspberry Pi OS with latest updates.

Any help is appreciated.

I hope it is not a faulty case problem...

kmarien, 2021-07-14:
I found the problem. It was a fried gpio pin 14
I fried the pin when I put the wiring block on the pins but started with pin 3 and 4 instead of 1 and 2 and then booted my pi.

I rewired gpio14 to gpio15 and adjusted the script for that pin and now everything works as a charm!

Close please!

jrkmtf, 2023-07-07:
Seems I'm having a similar issue - at least the behavior is the same. No LED when the safe shutdown switch is ON but it works fine when the switch is OFF. How did you edit the script? Did you edit a file on your pi after you installed the script?

kmarien, 2023-07-07:
I changed in /opt/Retroflag/Safeshutdownscript pin 14 to pin 15

jrkmtf, 2023-07-07:
Thank you! I see now where it references that file in install.sh. I'll give that a try, might help others in another issue.