RetroFlag/retroflag-picase #126: Skript wont install

renegade2k82, 2021-10-04:
For the life of me i cannot get the skript to run on my ubuntu and this really sucks because i like the case alot.

Michael-Zinn, 2021-10-04:
@renegade2k82 You should not run the script on your Ubuntu. You should run it on RetroPie or Recalbox on the pi (zero).
If you are running Ubuntu on a pi (zero) in a retroflag case, you will have to rewrite the script for Ubuntu yourself.

renegade2k82, 2021-10-05:
I have a pi 4b 8gb.so are you saying i cant use the safe shutdown feature on ubuntu?Do i have to have that script to use the buttons on the case?

zero4281, 2021-10-05:
> I have a pi 4b 8gb.so are you saying i cant use the safe shutdown feature on ubuntu?Do i have to have that script to use the buttons on the case?

The script has to be run on the RaspberryPi with RetroPie installed on the SD Card.  I don't think that Ubuntu is supported.

renegade2k82, 2021-10-05:
Oh ok so if i run the script with retropie then switch back to ubuntu will the skirpt still take effect or do i have to have retropie installed forever for the skirpt to work?

zero4281, 2021-10-05:
If you replace the operating system then the script won't be installed anymore.  You have to use RetroPie or customize the script to run with Ubuntu on your own.

renegade2k82, 2021-10-05:
ok well can  i just set the safe shutdown switch to the off position and still use the case buttons to turn it on and off and to restart?

zero4281, 2021-10-05:
> ok well can i just set the safe shutdown switch to the off position and still use the case buttons to turn it on and off and to restart?

That will do an unsafe shutdown that could corrupt the SD card.

renegade2k82, 2021-10-05:
Well i guess i will be returning these retroflag cases then its a huge let down that you have to use the safe shutdown just to not corrupt anything and also im disappointed thats its support is very slim.not everyone uses retropie and the other ones.

zero4281, 2021-10-05:
P.S. The issue with corrupting the SD card isn't because of the case.  The Raspberry Pi is always subject to that issue if you pull the plug on it.

renegade2k82, 2021-10-05:
Yeah i have used retropie before i will say it is very nice and runs well but i only want to use my pi to host minecraft servers so thats why i hoped there was some way for the safe shutdown to work on ubuntu.would this work https://github.com/crcerror/retroflag-picase/pull/7/commits/128b311a1dbd9cb27bedf2abbf545d3a6e9042d2

zero4281, 2021-10-05:
> Yeah i have used retropie before i will say it is very nice and runs well but i only want to use my pi to host minecraft servers so thats why i hoped there was some way for the safe shutdown to work on ubuntu.would this work [crcerror@128b311](https://github.com/crcerror/retroflag-picase/commit/128b311a1dbd9cb27bedf2abbf545d3a6e9042d2)

It's worth a shot dude.  It's a nice case, but it is a little bit of a DIY project.

alanbchristie, 2021-12-10:
If you want to run on Ubuntu you can. I'm using the 64-bit 21.04 on an 8Gi RPi 4 but had a number of problems. The shutdown script is broken anyway (#125) but that's an easy fix. In the end I created some notes for my installation (just in case I needed to do it again). Don't know if they'd be of any use...

https://gist.github.com/alanbchristie/a421cb9568ac9611436e16dee75e589b