RetroFlag/retroflag-picase #130: It is possible install script via SSH on Recalbox?

feliguez, 2022-03-29:
Hello everybody:
That's my point, actually I have a Megapi case and I wonder if it's possible to install this script via SSH on Recalbox 7.1.1, then if this is possible where should I run the script? 
I really appreciate your help, cause I don't have a keyboard to follow the defaults steps.

Regards!

feliguez, 2022-03-29:
thanks for you quick reply!
I already have access via ssh on my recalbox system. Now I just need to know if I can run the safe shutdown script and where. By default on Recalbox ssh access the directory is /recalbox/share/system, so, can I run the recalbox script on this path on shooud be in different one?

zero4281, 2022-03-29:
It might handle all the paths itself.

feliguez, 2022-03-29:
it works like a charm on the default path! thank you @zero4281