RetroFlag/retroflag-picase #133: Safe-shutdown Not working

davefromcamp, 2022-07-08:
My setup is a Rasp Pi 4 8GB ram that boots to RetroPie from a SSD (the cartridge in the case). Initially the safe shutdown was working great, not sure when or why it stopped. I easily could have changed a setting (not the internal switch) that I am unaware of. I did try the options listed here:

https://github.com/RetroFlag/retroflag-picase/issues/90

With no help.

I have up to date firmware.

I uninstalled and installed the other 'fork', didn't work either. Uninstalled the fork, and tried again with this script.

The power LED is on when the device is on. If I press reset the device resets. If I press power it just 'Shuts off' without running the proper shutdown. I have made a retropie system before, where the script ran perfectly, but ran off the SD card (card was corrupted after 1.5 years and lost everything - all my SAVES!!!) so I rebuilt but this time running off the SSD. Does that have something to do with this?

I also made a retropie Nespi4 case setup with 4gb ram for a friend and the same thing, The safe shutdown script doesn't work.

yoyojacky, 2023-08-15:
just modidy the /opt/RecalFlag/SafeShutdown.py  file 
modify the  os.system('shutdown -r now")  in poweroff function to os.system('shutdown -h now‘） and then save it and reboot your raspberry pi will solve this issue. it seems like they have made a mistake to put -r reboot argument .  -h means halt

fb64, 2023-09-20:
> just modidy the /opt/RecalFlag/SafeShutdown.py file modify the os.system('shutdown -r now") in poweroff function to os.system('shutdown -h now‘） and then save it and reboot your raspberry pi will solve this issue. it seems like they have made a mistake to put -r reboot argument . -h means halt

Thanks @yoyojacky , it works for me a Raspberry 3 B+ on Raspberry Pi OS (Bullseye)