RetroFlag/retroflag-picase #143: Raspberry Pi OS 12 bookworm kernel 6.1.69

djayor, 2024-03-12:
Hi, I'm using a Raspberry Pi 4 in a NESPI 4 Case.
I did a clean install of Raspberry Pi OS bookworm 12, and installed the script for RetroPie (this script works perfectly for buster, bullseye and bookworm in the last few years).

But in the moment that I upgrade the kernel 6.1.21 to 6.1.69 on bookworm, the script stop work.
The power button still allow to turn on, but cut the power immediately pressing again to turn it off. Reset doesn't do anything.

I already reinstalled the script, but no luck.
To be sure did another clean install and I just apply the commands:

$ sudo apt-get dist-upgrade -y
$ sudo apt-get --purge -y autoremove

Thanks,

chrisdreams13, 2025-01-22:
I tried to install bookworm from raspberry Pi Imager, the script is not working anymore, I have to downgrade the image to bullseye, does anybody has a solution to this?

djayor, 2025-04-19:
I found a solution provided by the fantastic work of the user @LPSeg13, that is documented here:
https://forums.raspberrypi.com/viewtopic.php?t=374127&start=25

After the installation to fix this issue do the steps:
$ cd /opt/RetroFlag/
$ sudo mv SafeShutdown.py SafeShutdown.py_old
$ sudo nano SafeShutdown.py

Past the following code:

```
import RPi.GPIO as GPIO
import os
import time
from multiprocessing import Process

# initialize pins
powerPin = 3 #pin 5
ledPin = 14 #TXD
resetPin = 2 #pin 13
powerenPin = 4 #pin 5

# initialize GPIO settings
def init():
	GPIO.setwarnings(False)
	GPIO.setmode(GPIO.BCM)
	GPIO.setup(powerPin, GPIO.IN, pull_up_down=GPIO.PUD_UP)
	GPIO.setup(resetPin, GPIO.IN, pull_up_down=GPIO.PUD_UP)
	GPIO.setup(ledPin, GPIO.OUT)
	GPIO.output(ledPin, GPIO.HIGH)
	GPIO.setup(powerenPin, GPIO.OUT)
	GPIO.output(powerenPin, GPIO.HIGH)

# waits for user to hold button up to 1 second before issuing poweroff command
def poweroff():
	while True:
		# self.assertEqual(GPIO.input(powerPin), GPIO.LOW)
		# GPIO.wait_for_edge(powerPin, GPIO.FALLING)
		while GPIO.input(powerPin) != GPIO.LOW:
                    time.sleep(0.2)
		os.system("sudo killall emulationstation")
		os.system("sudo killall emulationstatio") #RetroPie 4.6
		os.system("sudo sleep 5s")
		os.system("sudo shutdown -P now")

# blinks the LED to signal button being pushed
def ledBlink():
	while True:
		GPIO.output(ledPin, GPIO.HIGH)
		# self.assertEqual(GPIO.input(powerPin), GPIO.LOW)
		# GPIO.wait_for_edge(powerPin, GPIO.FALLING)
		while GPIO.input(powerPin) != GPIO.LOW:
                    time.sleep(0.2)
		start = time.time()
		while GPIO.input(powerPin) == GPIO.LOW:
			GPIO.output(ledPin, GPIO.LOW)
			time.sleep(0.2)
			GPIO.output(ledPin, GPIO.HIGH)
			time.sleep(0.2)

# resets the pi
def reset():
	while True:
		# self.assertEqual(GPIO.input(resetPin), GPIO.LOW)
		# GPIO.wait_for_edge(resetPin, GPIO.FALLING)
		while GPIO.input(resetPin) != GPIO.LOW:
                    time.sleep(0.2)
		os.system("sudo killall emulationstation")
		os.system("sudo killall emulationstatio") # RetroPie 4.6
		os.system("sudo sleep 5s")
		os.system("sudo shutdown -r now")

if __name__ == "__main__":
	# initialize GPIO settings
	init()
	# create a multiprocessing.Process instance for each function to enable parallelism 
	powerProcess = Process(target = poweroff)
	powerProcess.start()
	ledProcess = Process(target = ledBlink)
	ledProcess.start()
	resetProcess = Process(target = reset)
	resetProcess.start()

	powerProcess.join()
	ledProcess.join()
	resetProcess.join()

	GPIO.cleanup()
```

Save (Control+O and Control+X), can be tested before the restart using:
$ python3 SafeShutdown.py 

(I left all the steps, because not everyone who will encounter this difficulty may know how to use the command line.)

closed 2025-04-19 by djayor